sllurp/sllurp #151: Could not start session on reader: Failed (a Client initiated connection already exists)

aymeri07, 2020-02-18:
Hi everybody,

i use speedwayRevolution and i program with python3. I discover a sllurp library. But i have problems with this little program to read RFID tag. It's possible help me ?

`from sllurp import llrp
from twisted.internet import reactor
import logging
logging.getLogger().setLevel(logging.INFO)

def cb (tagReport):
    tags = tagReport.msgdict['RO_ACCESS_REPORT']['TagReportData']
    print('tags:', tags)

factory = llrp.LLRPClientFactory()
factory.addTagReportCallback(cb)
reactor.connectTCP('192.168.5.10', llrp.LLRP_PORT, factory)
reactor.run()`

My error is Could not start session on reader: Failed (a Client initiated connection already exists), but i don't know this error ? Have you an example program to read RFID tag with two antennas ?

thank you so much.

ransford, 2020-02-19:
This is due to LLRP sessions' already being active when you try to connect.
You can't have two clients at once controlling a reader. It may be that a
previous session is stuck somehow after an awkward termination. I recommend
running `sllurp reset 1.2.3.4` where 1.2.3.4 is the IP address of your
reader.

aymeri07, 2020-02-19:
thank you ransford,
I do `sllurp reset 192.168.5.10` but i have this error message : 
2020-02-19 10:12:37,016 sllurp.llrp: INFO: connecting to 192.168.5.10:5084...
2020-02-19 10:12:37,017 sllurp.llrp: INFO: using antennas: (1,)
2020-02-19 10:12:37,018 sllurp.llrp: INFO: transmit power: {1: 0}
2020-02-19 10:12:37,019 sllurp.llrp: INFO: connected to ('192.168.5.10', 5084) (192.168.5.10:5084)
2020-02-19 10:12:37,341 sllurp.llrp: CRITICAL: Could not start session on reader: Failed (a Client initiated connection already exists)
2020-02-19 10:12:37,343 sllurp.llrp: INFO: lost connection: Connection was closed cleanly.

it's crazy, because i just upgrade speedway reader with upg files (SpeedwayConnect-2.8.0.0.upg) directly on http://<ip adress>/cgi-bin/index.cgi and i change ip adress of my reader.

that's all, have you the same problem ?

thnak you

ransford, 2020-02-19:
Yes, you can change the active set of antennas; see `sllurp inventory --help` for the `inventory` command's arguments and see `sllurp/verb/inventory.py` for the implementation of that command.

closed 2020-02-19 by ransford

furiiibond, 2023-01-23:
I still have the problem : 
```
C:\Users\lapierrj>sllurp reset 192.168.1.3
2023-01-23 09:26:26,862 sllurp.llrp: INFO: connecting to 192.168.1.3:5084...
2023-01-23 09:26:26,862 sllurp.llrp: INFO: using antennas: (1,)
2023-01-23 09:26:26,862 sllurp.llrp: INFO: transmit power: {1: 0}
2023-01-23 09:26:26,862 sllurp.llrp: INFO: connected to ('192.168.1.3', 5084) (192.168.1.3:5084)
2023-01-23 09:26:27,216 sllurp.llrp: CRITICAL: Could not start session on reader: Failed (a Client initiated connection already exists)
2023-01-23 09:26:27,216 sllurp.llrp: INFO: lost connection: Connection was closed cleanly
```