sllurp/sllurp #153: politeShutdown doesn't end after disconnect with the RFID reader. Tried #116 and #136

edmond1992, 2020-04-17:
I got a similar problem to #116 and #136's solution doesn't work for me.
The shutdown task ran, but it never return to main despite it said STATE_DISCONNECTED and Connection was closed cleanly.
Reader is speedway R420

[debug.txt](https://github.com/ransford/sllurp/files/4494190/debug.txt)
```
#Some other stuff
SCAN_TIME = 5.0
def shutdown(factory):
    print('Shutting down')
    print(factory.politeShutdown())
    return 0

def main():
    factory = llrp.LLRPClientFactory(**readerargs)
    factory.addTagReportCallback(report)
    reactor.connectTCP(READER_IP_ADDRESS, llrp.LLRP_PORT, factory, timeout=3)
    reactor.addSystemEventTrigger('before', 'shutdown', shutdown, factory)
    # https://twistedmatrix.com/documents/current/core/howto/time.html
    reactor.callLater(SCAN_TIME, shutdown, factory)
    reactor.run()
    print("Back")
    return 0
```

fviard, 2020-04-17:
Hi @edmond1992, If possible, let me suggest you to switch to use my branch instead, that is not dependent anymore  on twisted:
https://github.com/ransford/sllurp/tree/fviard-develop-v2
You have the correct usage example updated in this branch also.

If you want, you can have more details on the inception of this branch here:
https://github.com/ransford/sllurp/issues/106

edmond1992, 2020-04-17:
> Hi @edmond1992, If possible, let me suggest you to switch to use my branch instead, that is not dependent anymore on twisted:
> https://github.com/ransford/sllurp/tree/fviard-develop-v2
> You have the [...]

So, it is a twisted related issue?
One more question, does your branch support Impinj extensions? I need it for the phase data.

fviard, 2020-04-17:
I can't swear it, but i think that it is possible.
Yes for impinj extension.
The v2 branch is the same features that you already have, but completely
reworked to be more efficient (x1000), needing less dependencies, to
support a lot more things and also have lots of bugs fixed on the way.

papapel, 2020-04-21:
Hello @edmond1992, 

https://github.com/ransford/sllurp/tree/fviard-develop-v2 works well. I have used it for 3 months and I never had a crash. It is very reliable.

closed 2021-01-24 by ransford