sllurp/sllurp #155: Step by step tutorial for beginners

withoutcode, 2020-06-02:
Hi
I am a young Python developer and have been starting to work with RFID technology in Python.
Maybe my problem is very simple for you, but it is my problem and if you can help me, I appreciate it. 
I have an Impinj R-420 reader and would like to read tags data with python. I already did it by Octance Java SDK from the Impinj company.
I was able to install sllurp and connect to the reader, but I want to

- - See the tags result real-time
- - Save the result in a .csv file
- - Start and stop the reader manually.
- -Transfer data to a database

Can you help me how can I do that? 
Is there any tutorial for a beginner like me? Because when I read the comments, somebody mentioned " add this line to your file" but it is difficult for me to understand what file and what line.
Thank you in advance

thijmenketel, 2020-06-27:
See my last comment under #149, it might help you if you can get it to work.

ransford, 2020-06-30:
Hi @withoutcode, I would suggest the following.

1. Get the "inventory" command to work with your RFID reader. It sounds like you already did this. The simplest way is to run `sllurp inventory 1.2.3.4` where 1.2.3.4 is the IP address of your RFID reader. This will confirm that your reader is compatible with sllurp.
2. Copy the contents of the program at https://github.com/ransford/sllurp#reader-api to a new file called, say, `myapp.py`. This will give you a very basic program to experiment with.
3. Experiment with editing the `cb()` ("callback") function of `myapp.py` until it does what you want. You would put database queries and CSV printing in this function, for example.
4. `myapp.py` is the file you will run. A sllurp application can be very short and self contained. You can ignore most of the functionality of sllurp if you just want to read RFID tags and start and stop.

Most applications that rely on sllurp do something similar: most of the interesting stuff is in the callback function.

The simplest way to stop the reader is by calling `stopPolitely()`. You can see a small example in `sllurp/verb/reset.py`.

Good luck with your first application!

closed 2021-01-24 by ransford