sllurp/sllurp #157: How to use " access write " command?

ZIQI-code, 2020-07-31:
I am a beginner in using sllurp. I am using Impinj Speedway Revolution R420 has my reader and EM4325 to be my tag. I want to modify the memory from something else to \x0000. For example I would write `sllurp access -mb 3 -c 1 -wp 240 -w 1 `in cmd and then when cmd shows `2020-07-31 14:20:46,609 sllurp.llrp: INFO: starting inventory`, I would write \x0000. However, if I read it, cmd returns `'ReadData': b'\\x'`. I want to store the words or numbers I typed in directly into the memory. However, it seems that the `sllurp access write` command will always convert the first two characters into ASCii and store that two characters. Can someone please help me with this issue. I just want to store the binary number or the hex number I typed in directly into the memory. Thank you very much!

ransford, 2020-08-06:
Hello! The `access` command is designed to read binary data from standard input. If you can put exactly the bytes you want into a file and provide that file's contents to `sllurp access` on stdin, you should see the write succeed with the bytes you provided. The words of your write are two bytes each, so make sure you provide an even number of bytes.

Hope that helps!

ZIQI-code, 2020-08-11:
I kind of understand. But how to provide the file's contents to `sllurp access` on stdin? Can you explain it more or offer an example? Thank you very much!!!

ransford, 2020-08-12:
Using a pipe! I'd refer you [here](https://en.wikipedia.org/wiki/Pipeline_(Unix)) or [here](https://drawings.jvns.ca/pipes/) for starters.

closed 2021-01-24 by ransford