sllurp/sllurp #163: Error while trying to read WISP-5 with SLLURP library

keyrons-hub, 2021-06-16:
Hello all, 
I have been trying to connect a WISP 5 reader using the instructions available on the WISP wiki - https://sites.google.com/uw.edu/wisp-wiki/wisp5. However, when I try to run the command "sllurp inventory -n 1 -T 7140 -M WISP5 –mode-identifier 0 [ipaddr]" I get an error that says "Error: No such option: -M". 

Looking at the SLLURP inventory, there does not seem to be a command for -M, so I'm not sure where that came from (the WISP is quite a few years old). When I remove the "-M" from the command I get a host of errors located in the defer.py, base.py, log.py, and other SLLURP python files. It also says that the connection is refused with the reader. 

I should note that this reader works with other RFID tags, so not entirely sure what the issue is with this command trying to get it to work on the WISP. Any help is appreciated!

ransford, 2021-06-18:
The `-M` mode selection switch moved to `--mode-identifier`; I believe `--mode-identifier 0` should do what you're looking for. [This explains](https://github.com/ransford/sllurp/issues/114#issuecomment-450518892). Good luck!

closed 2021-06-18 by ransford

keyrons-hub, 2021-06-21:
Hello, I am running into more errors when using the --mode-identifier 0
command and trying to connect to a WISP 5. The error seems to lie in an
inability to connect to the WISP, or perhaps the reader as well, I'm not
entirely sure. In the sllurp/llrp.py file, it says "connecting to
--mode-identifier:5084", which I believe is the WISP but I'm not sure. I
know you don't deal with the WISP, so I apologize for all the questions. I
have attached the error messages below.

Thanks for any help you can provide,
Kieran Humphreys

ransford, 2021-07-02:
It looks like you somehow convinced `inventory` that you wanted to connect to a _hostname_ called `--mode-identifier`, something like
```shell
## note the extra ' -- '
$ sllurp inventory -- --mode-identifier 192.168.1.1
2021-07-02 16:45:49,331 sllurp.verb.inventory: INFO: Antenna map: {'--mode-identifier': {'1': 'Antenna 1'}, '192.168.1.1': {'1': 'Antenna 1'}}
2021-07-02 16:45:49,331 sllurp.llrp: INFO: connecting to --mode-identifier:5084...
```

You probably figured it out!