sllurp/sllurp #164: Help needed for using a FX7500 with two antennas

TonySpegel, 2021-07-09:
Hello, we would like to use your project to be able to read RFID tags via two antennas connected to a FX7500 from Zebra. We are using your minimal example so far in a slightly modified form and the first thing we noticed is that the first time we run the script, nothing seems to be logged but when we run it a second time, tags are logged almost immediately. The second thing we noticed is that even though we have two antennas configured, the log never lists tags with an antenna ID other than 1. Both antennas are active and via the manufacturer's interface they can scan something. Here is our code so far and a sample output of the scan.

```python
from sllurp import llrp
from twisted.internet import reactor
import logging

sllurp_logger = logging.getLogger('sllurp')
sllurp_logger.setLevel(logging.INFO)
sllurp_logger.addHandler(logging.StreamHandler())

def cb(tagReport):
    tags = tagReport.msgdict['RO_ACCESS_REPORT']['TagReportData']
    for tag in tags:
        print(tag)

factory = llrp.LLRPClientFactory(
    antenna_dict={'10.120.1.100': {'1': 'ant1', '2': 'ant2'}},
    tag_content_selector={
        'EnableAntennaID': True,
        'EnableChannelIndex': True,
        'EnablePeakRSSI': True,
        'EnableFirstSeenTimestamp': True,
        'EnableLastSeenTimestamp': True,
        'EnableTagSeenCount': True,
        'EnableAccessSpecID': True,
    }
)

factory.addTagReportCallback(cb)

reactor.connectTCP('10.120.1.100', llrp.LLRP_PORT, factory)
reactor.run()
```
```bash
 ~/Dev/work/slurp  python3 slurp.py                                                                                                                                                        ✔  slurp-_tkCAM08   15:35:34
connecting to 10.120.1.100:5084...
will reset reader state on connect
will start inventory on connect
using antennas: [1, 2]
transmit power: {1: 0, 2: 0}
connected to ('10.120.1.100', 5084) (10.120.1.100:5084)
using reader mode: None
stopping politely
Impinj search mode? None
starting inventory
^Clost connection: Connection to the other side was lost in a non-clean fashion: Connection lost.

 ~/Dev/work/slurp  python3 slurp.py                                                                                                                                                   ✔  6s  slurp-_tkCAM08   15:35:45
connecting to 10.120.1.100:5084...
will reset reader state on connect
will start inventory on connect
using antennas: [1, 2]
transmit power: {1: 0, 2: 0}
connected to ('10.120.1.100', 5084) (10.120.1.100:5084)
using reader mode: None
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'5468576d5a7134743777217a25432646'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-56,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741204148,), 'LastSeenTimestampUTC': (1625837741204148,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'566b597033733676397924423f452848'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-70,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741208293,), 'LastSeenTimestampUTC': (1625837741208293,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPC-96': b'e2806894000040012e931019', 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-63,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741212165,), 'LastSeenTimestampUTC': (1625837741212165,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'78214125442a472d4b6150645367566b'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-62,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741216989,), 'LastSeenTimestampUTC': (1625837741216989,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'3272357538782f413f4428472b4b6250'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-59,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741221735,), 'LastSeenTimestampUTC': (1625837741221735,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPC-96': b'e2806894000050012e948c19', 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-54,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741227329,), 'LastSeenTimestampUTC': (1625837741227329,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'34743777217a25432a462d4a614e6452'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-63,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741231730,), 'LastSeenTimestampUTC': (1625837741231730,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'4d6251655468576d5a7133743677397a'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-58,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741236008,), 'LastSeenTimestampUTC': (1625837741236008,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'556b58703273357638792f413f442847'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-66,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741240267,), 'LastSeenTimestampUTC': (1625837741240267,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'51665468576d5a7134743777217a2543'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-66,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741244522,), 'LastSeenTimestampUTC': (1625837741244522,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'264528482b4d6251655468576d5a7134'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-59,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741248791,), 'LastSeenTimestampUTC': (1625837741248791,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'7134743777397a24432646294a404e63'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-56,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741253111,), 'LastSeenTimestampUTC': (1625837741253111,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'5970337336763979244226452948404d'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-70,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741257405,), 'LastSeenTimestampUTC': (1625837741257405,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'2646294a404e635266556a586e327235'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-69,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741262596,), 'LastSeenTimestampUTC': (1625837741262596,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPC-96': b'e2806894000040012e948819', 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-62,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741268964,), 'LastSeenTimestampUTC': (1625837741268964,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'576d5a7134743777217a25432a46294a'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-66,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741277516,), 'LastSeenTimestampUTC': (1625837741277516,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'3f4428472b4b6250645367566b597033'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-70,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741286052,), 'LastSeenTimestampUTC': (1625837741286052,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'59703373357638792f423f4528482b4d'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-68,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741293641,), 'LastSeenTimestampUTC': (1625837741293641,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'6e5a7234753777217a25432a462d4a61'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-69,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741297957,), 'LastSeenTimestampUTC': (1625837741297957,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'294a404e635266556a586e3272357538'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-67,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741303073,), 'LastSeenTimestampUTC': (1625837741303073,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
[... cut ...] ': (1625837741307306,), 'LastSeenTimestampUTC': (1625837741307306,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
{'EPCData': {'EPCLengthBits': 128, 'EPC': b'6250655368566d597133743677397a24'}, 'ROSpecID': (1,), 'AntennaID': (1,), 'PeakRSSI': (-65,), 'ChannelIndex': (3,), 'FirstSeenTimestampUTC': (1625837741312183,), 'LastSeenTimestampUTC': (1625837741312183,), 'TagSeenCount': (1,), 'AccessSpecID': (0,)}
stopping politely
Impinj search mode? None
starting inventory
```

Thank you for your help :)

ransford, 2021-07-12:
Does `sllurp inventory -a 0 -n 1 10.120.1.100` show tags from both antennas? If so, you can add a `--debug` switch (`sllurp --debug inventory -a 0 -n -1 ...`) and see the messages to the reader that make it do the right thing.

The next thing I'd try in your own project is to turn sllurp's log level to `DEBUG` and inspect the output. See the `Logging` section of the README for an example of how to do that. If the behavior you elicit is different from the behavior of the `inventory` script, you can look probably find the answer in the debugging output.

There's nothing obviously amiss with your code. What happens if you physically disconnect Antenna 1?

Yury-tech, 2021-08-13:
Hello! I have similar problem, but in my case fx7500 read only 3 antennas (4 connected). Maybe my code can help you:

`factory_args = dict(
        start_first=True,
        duration=1,
        antenna_dict=antmap,
        tx_power=0,
        session=0,
        tag_population=4,
        start_inventory=True,
        disconnect_when_done=False,
        reconnect=True,
        tag_content_selector={
            'EnableROSpecID': False,
            'EnableSpecIndex': False,
            'EnableInventoryParameterSpecID': False,
            'EnableAntennaID': True,
            'EnableChannelIndex': False,
            'EnablePeakRSSI': True,
            'EnableFirstSeenTimestamp': False,
            'EnableLastSeenTimestamp': True,
            'EnableTagSeenCount': True,
            'EnableAccessSpecID': False
        }
    )

def cb(tagReport):
    try:
        tags = tagReport.msgdict['RO_ACCESS_REPORT']['TagReportData']
        name = tagReport.peername
        print('reader: {} tags: {} '.format(name,tags))
        readerIP = name[0]
        if len(tags):
            for tag in tags:
                EPC = tag.get('EPC-96',None)
                AntennaID = tag['AntennaID'][0]
                LastSeenTimestampUTC = tag['LastSeenTimestampUTC'][0]
                LastSeenTimestampUTC=datetime.fromtimestamp(LastSeenTimestampUTC/1000000)
                TagSeenCount = tag['TagSeenCount'][0]
                PeakRSSI = tag['PeakRSSI'][0]
                InsertTag(EPC, AntennaID, TagSeenCount, LastSeenTimestampUTC, PeakRSSI, readerIP)
    except Exception as err:
        print('Error while tagReport (function write mysql): {}'.format(err))
factory = llrp.LLRPClientFactory(**factory_args)
factory.addTagReportCallback(cb)

for host in ReaderHostIPs:
    if ':' in host:
        host, port = host.split(':', 1)
        port = int(port)
    else:
        port = 5084
    conn = reactor.connectTCP(host, llrp.LLRP_PORT, factory, timeout=3)
reactor.run()
`

My ant map look like:
antmap = {'192.168.1.31:5084': {'1': 'ant1','2':'ant2','3':'ant3','4':'ant4'}. When i cut 1,2 or 3 antenna it read :).
Seems like Report bug?

Yury-tech, 2021-08-13:
Addition: when change antmap like antmap = {'192.168.1.31:5084': {'1': 'ant1','3':'ant3','4':'ant4','2':'ant2'}} tag report give tags from: 1,3,4 antennas, but now from two. Last argument ignored.

closed 2022-04-24 by ransford