sllurp/sllurp #167: Output to insert mysql

phuriphak, 2021-12-11:
Hello, i'm trying to use this project for implement for Insert to MySQL anyway , 
 I don't know how to start it . from where.

could you help me for this request. 

Thank you so much.

ransford, 2021-12-13:
If you want to insert directly from your application to MySQL, try creating your own callback function (like [the example](https://github.com/ransford/sllurp#reader-api) in the README) that calls the insert function. If your database can't keep up with the tag rate, try using the `report_every_n_tags=N` argument when creating your `LLRPClient` to reduce the reports by a factor of N.

closed 2021-12-13 by ransford

phuriphak, 2021-12-14:
Thank you for your anwser , anyway When program running i can use command for running or not. I try to edit for Inventory in .verb  but when running "sllurp inventory -s 0 -t 1 169.254.10.1" I can't change result or wording in logger. 

could you tell me for where .verb/inventory that's edit and can use for "sllurp inventory"

Thanks.

phuriphak, 2021-12-14:
More question now I make myapp.py for test -> result was out , in my purpose need to continues for receive tag. need to add coding from where. could you explain more for continue read tag.

ransford, 2021-12-16:
You can write your own Python program, and then you won't have to interact with the `verb/inventory` stuff at all. If you copy the block of code in the README ([here](https://github.com/ransford/sllurp/#reader-api)), you'll get a simple working client that you can customize to fit your needs.