sllurp/sllurp #168: Cant connect using basic sllurp inventory {ip} - FX7500

stifferdoroskevich, 2022-01-24:
Hi Everyone!

Im trying to connect using sllurp inventory {ip}.

```code
sllurp inventory -s 0 -t 1 192.168.88.253
2022-01-24 20:41:09,023 sllurp.verb.inventory: INFO: Antenna map: {'192.168.88.253': {'1': 'Antenna 1'}}
2022-01-24 20:41:09,023 sllurp.llrp: INFO: connecting to 192.168.88.253:5084...
2022-01-24 20:41:09,027 sllurp.llrp: INFO: will reset reader state on connect
2022-01-24 20:41:09,027 sllurp.llrp: INFO: will start inventory on connect
2022-01-24 20:41:09,028 sllurp.llrp: INFO: using antennas: [1]
2022-01-24 20:41:09,028 sllurp.llrp: INFO: transmit power: {1: 0}
2022-01-24 20:41:09,029 sllurp.llrp: INFO: connected to ('192.168.88.253', 5084) (192.168.88.253:5084)
2022-01-24 20:41:10,610 sllurp.llrp: INFO: using reader mode: None
.
.

  File "c:\users\stiffer doroskevich\development\python\ransford_sllurp\.venv\lib\site-packages\sllurp\llrp_decoder.py", line 52, in decode_tve_parameter
    param_name, param_fmt = ext_param_formats[subtype]
builtins.KeyError: 709

2022-01-24 20:41:10,742 sllurp.llrp: INFO: lost connection: 709
```

If I print the ext_param_formats
This is the result:
```
{56: ('ImpinjPhase', '!H'), 57: ('ImpinjPeakRSSI', '!h'), 68: ('RFDopplerFrequency', '!h')}
```

Anyone have an idea in how to proceed?

PS: I Can connect using the web interface directly in the ip.
PS2: This could be a Routing problem in the network?

Thanks in advance!

ransford, 2022-01-25:
Your reader is sending a custom parameter that sllurp doesn't know how to interpret. You might find something in your reader's manual explaining the `709` custom parameter and its argument, then add your findings to the `ext_param_formats` dictionary. For example, if the parameter with type 709 includes an unsigned 2-byte integer, you'd add something like `509: ('MyReaderSpecialParam', '!H')` to `ext_param_formats`.

stifferdoroskevich, 2022-01-25:
Thanks Ransford!

I created a basic socket to verify if I can connect. 
This is the response when I try to .decode():

```
KeyError: b'\x04?\x00\x00\x00 \x00\x00\x00\x05\x00\xf6\x00\x16\x00\x80\x00\x0c\x00\x05\xd6i\x8b{\xfa,\x01\x00\x00\x06\x00\x00'
```
and if a print the result, it gives me an array of integers (32 items):
```
[4, 63, 0, 0, 0, 32, 0, 0, 0, 8, 0, 246, 0, 22, 0, 128, 0, 12, 0, 5, 214, 105, 157, 26, 133, 160, 1, 0, 0, 6, 0, 0]
```

Thanks for the support, I will look at the manual to find that code, also if I cant find, will send an email to Zebra.

I will keep this issue updated.

stifferdoroskevich, 2022-02-02:
Found the problem!

Was a new parameter by Zebra in the new firmware.
I will make more tests first, and after that will share here (create a PR to include in this lib).

ransford, 2022-02-02:
Interesting. Thanks for looking! PRs welcome.

closed 2022-03-28 by ransford

idzabdan22, 2022-04-24:
@stifferdoroskevich hai, i just have the same problem as you, can you tell me how to solve this error 709 while reading the tags? im literally stuck all night trying to figure it out, would be a big thanks if you can tell me the solution

stifferdoroskevich, 2022-04-26:
hi ! I added a new parameter to the lib and is still working.

I will add that and do a PR to the lib tonight, ok?

idzabdan22, 2022-04-26:
Sounds great! I'll be waiting to see your PR soon, all thanks to you!

fviard, 2022-04-26:
@stifferdoroskevich Would be nice for you to do a PR.

By the way, if you can give a try and eventually also do your PR for the following branch it would be great:
https://github.com/sllurp/sllurp/tree/fviard-develop-v2

It should soon be merged in developer and become a new version.
Also, this new version is a lot more performant, like 1000 times faster, so that could be of interest to you :-)