sllurp/sllurp #169: Tags are not purged when tags are removed

ganeshasri, 2022-03-24:
When we read a set of tags & get the tag counts & disconnect from reader, and take out few tags from the set & read it again, it gives the same number of found tags. 
The count is not reducing, do we have to do anything explicitly to restart the count afresh?

ganeshasri, 2022-03-24:
When we run python code 2nd time it gives correct output.

ganeshasri, 2022-03-24:
We ran below code in command prompt and it works fine:
class ResetRFIDReader:

    def __init__(self, configFile: configparser.ConfigParser()):
        self.configFile = configFile
        ''' Load the logger configuration '''
        logging.config.fileConfig(fname=self.configFile['File']['LoggerConfig'], disable_existing_loggers=False)
        self.logger = logging.getLogger(__name__)
        self.iConnector = None

        self.factory = llrp.LLRPClientFactory(reset_on_connect=False,
                                              start_inventory=False)
        self.factory.addStateCallback(llrp.LLRPClient.STATE_CONNECTED, self.shutdownReader)

    def shutdownReader(self, proto):
        self.logger.info('shutdownReader RFID reader')
        return proto.stopPolitely(disconnect=True)

    def stopReactor(self):

        if not self.iConnector:
            try:
                self.connectTCP()
            except Exception as error:
                self.logger.error(Constants.rfidOpenConnectError + ', with error:' + str(error))

        try:
            if reactor.running:
                reactor.stop()
        except Exception as error:
            self.logger.error(Constants.rfidCloseConnectError + ', with error:' + str(error))

    def resetRFIDReader(self):

        factory = llrp.LLRPClientFactory(reset_on_connect=False,
                                         start_inventory=False)
        factory.addStateCallback(llrp.LLRPClient.STATE_CONNECTED, self.shutdownReader)

        try:

            self.iConnector = reactor.connectTCP(self.configFile["RFIDParamerer"]["Host"],
                                                 self.configFile.getint("RFIDParamerer", "Port"),
                                                 self.factory, self.configFile.getint("RFIDParamerer", "Timeout"))
        except Exception as error:
            self.logger.error(Constants.rfidReaderConnectError + "on RFID host:" +
                              self.configFile["RFIDParamerer"]["Host"] + ", on port:" +
                              str(self.configFile["RFIDParamerer"]["Port"]) + ", with error:" + str(error))
            raise RFIDError(Constants.rfidReaderConnectError + "on RFID host:" +
                            self.configFile["RFIDParamerer"]["Host"] + ", on port:" +
                            str(self.configFile["RFIDParamerer"]["Port"]) + ", with error:" + str(error))

        reactor.callLater(3, self.stopReactor)
        try:
            if reactor.running:
                reactor.stop()
            reactor.run(installSignalHandlers=False)
        except AlreadyCalledError as error:
            print('ResetRFIDReader AlreadyCalledError in reactor.run:', str(error))
        except Exception as error:
            print('ResetRFIDReader Exception in reactor.run:', str(error))

Please let us know if there are any alternate method to handle this.

ganeshasri, 2022-03-25:
It worked fine by changing the session id.
We used 0 to 4 session id in LLRPClientFactory constructor

ransford, 2022-03-28:
Glad you got it to work.

closed 2022-03-28 by ransford