sllurp/sllurp #172: how to implemnt try and except to read data from RFID using sllurp

ajsmack, 2022-05-16:
Hi,
I want  to implement to try and except in the following code if the connection is True then only I am able to read RFIDF tag .and I am not able to understand how slllrp Libary make connection between Reader and Tag 

so please help me to 

`from sllurp import llrp
from twisted.internet import reactor
import logging
logging.getLogger().setLevel(logging.INFO)

def cb (tagReport):
    tags = tagReport.msgdict['RO_ACCESS_REPORT']['TagReportData']
    print 'tags:', tags

factory = llrp.LLRPClientFactory()
factory.addTagReportCallback(cb)
reactor.connectTCP('myreader', llrp.LLRP_PORT, factory)
reactor.run()`

ransford, 2022-05-17:
In the code you pasted, `myreader` refers to the hostname or IP address of your RFID reader. The `connectTCP` call is common in Twisted applications ([doc](https://twistedmatrix.com/documents/current/core/howto/clients.html)). The `cb` function is a callback that Twisted will invoke when the reader sees tags. The `tagReport.msgdict` structure holds all of the data about the tags that the reader saw.

I would recommend adding more `print()` statements to inspect the various parts (e.g., `tagReport.msgdict`).

Good luck!

closed 2022-05-17 by ransford

ajsmack, 2022-05-17:
My code is working fine  I can read the tag but I want to apply to try and except  block and I do not understand how to do it in that code
whenever my reader is connected only that time I can read the tag otherwise it goes under except block which method do I use to check the current status of the connection object?