sllurp/sllurp #174: sllurp Could not start session on reade

furiiibond, 2023-01-23:
I still have the problem : 
```
C:\Users\lapierrj>sllurp reset 192.168.1.3
2023-01-23 09:26:26,862 sllurp.llrp: INFO: connecting to 192.168.1.3:5084...
2023-01-23 09:26:26,862 sllurp.llrp: INFO: using antennas: (1,)
2023-01-23 09:26:26,862 sllurp.llrp: INFO: transmit power: {1: 0}
2023-01-23 09:26:26,862 sllurp.llrp: INFO: connected to ('192.168.1.3', 5084) (192.168.1.3:5084)
2023-01-23 09:26:27,216 sllurp.llrp: CRITICAL: Could not start session on reader: Failed (a Client initiated connection already exists)
2023-01-23 09:26:27,216 sllurp.llrp: INFO: lost connection: Connection was closed cleanly
```

_Originally posted by @furiiibond in https://github.com/sllurp/sllurp/issues/151#issuecomment-1399968788_

BjornGrylls, 2023-05-10:
Hi
This error is because a client is already connected to the the reader. This can happen when you first start a client, then closes it without disposing of it.
An example:
\
First I will show you my current processes.
```
┌──(kali㉿kali)-[~]
└─$ ps
    PID TTY          TIME CMD
   1550 pts/0    00:00:00 zsh
   1645 pts/0    00:00:00 ps
```
We see my terminal, zsh. We also see the command that I am running, ps. This is the expected output.
\
I start a client.
```
┌──(kali㉿kali)-[~]
└─$ python sllurp/examples/examples/tornado/main.py 192.168.1.113 -X 10 -n 50
2023-05-10 11:08:59,612 sllurp.llrp: INFO: connecting to 192.168.1.113:5084...
2023-05-10 11:08:59,617 sllurp.llrp: INFO: will reset reader state on connect
2023-05-10 11:08:59,617 sllurp.llrp: INFO: will start inventory on connect
2023-05-10 11:08:59,617 sllurp.llrp: INFO: using antennas: [1]
2023-05-10 11:08:59,617 sllurp.llrp: INFO: transmit power: {1: 10}
2023-05-10 11:08:59,617 sllurp.llrp: INFO: connected to ('192.168.1.113', 5084) (192.168.1.113:5084)
2023-05-10 11:08:59,819 sllurp.llrp: INFO: using reader mode: None
2023-05-10 11:08:59,835 sllurp.llrp: INFO: stopping politely
Response:

Request:
"[{'EPC-96': b'393330393939393034353235', 'PeakRSSI': (-31,), 'LastSeenTimestampUTC': (1683708585317602,), 'TagSeenCount': (45,)}]"

2023-05-10 11:09:00,186 sllurp.llrp: INFO: Impinj search mode? None
2023-05-10 11:09:00,186 sllurp.llrp_proto: INFO: will report every ~N=50 tags
2023-05-10 11:09:00,186 sllurp.llrp: INFO: starting inventory
Response:

Request:
"[{'EPC-96': b'393330393939393034353235', 'PeakRSSI': (-31,), 'LastSeenTimestampUTC': (1683708586645859,), 'TagSeenCount': (50,)}]"

^C^Z
zsh: suspended  python sllurp/examples/examples/tornado/main.py 192.168.1.113 -X 10 -n 50
```
The client connects to the reader. I then suspend my client with CTRL+C, CTRL+Z
\
I start a new client.
```
┌──(kali㉿kali)-[~]
└─$ python sllurp/examples/examples/tornado/main.py 192.168.1.113 -X 10 -n 50
2023-05-10 11:09:07,845 sllurp.llrp: INFO: connecting to 192.168.1.113:5084...
2023-05-10 11:09:07,846 sllurp.llrp: INFO: will reset reader state on connect
2023-05-10 11:09:07,846 sllurp.llrp: INFO: will start inventory on connect
2023-05-10 11:09:07,846 sllurp.llrp: INFO: using antennas: [1]
2023-05-10 11:09:07,846 sllurp.llrp: INFO: transmit power: {1: 10}
2023-05-10 11:09:07,846 sllurp.llrp: INFO: connected to ('192.168.1.113', 5084) (192.168.1.113:5084)
2023-05-10 11:09:08,146 sllurp.llrp: CRITICAL: Could not start session on reader: Failed (a Client initiated connection already exists)
2023-05-10 11:09:08,146 sllurp.llrp: INFO: lost connection: Connection was closed cleanly.
^C                                                                                                                                                            
```
But I get the same error as you.
\
I check my processes.
```
┌──(kali㉿kali)-[~]
└─$ ps
    PID TTY          TIME CMD
   1550 pts/0    00:00:00 zsh
   1654 pts/0    00:00:01 python
   1706 pts/0    00:00:00 ps
```
I have a python process running. This is the first client I created. It is still running in the background.
\
I stop the python process.
```
┌──(kali㉿kali)-[~]
└─$ kill -9 1654

[1]  + killed     python sllurp/examples/examples/tornado/main.py 192.168.1.113 -X 10 -n 50
```
\
I start a new client.
```
┌──(kali㉿kali)-[~]
└─$ python sllurp/examples/examples/tornado/main.py 192.168.1.113 -X 10 -n 50
2023-05-10 11:09:34,793 sllurp.llrp: INFO: connecting to 192.168.1.113:5084...
2023-05-10 11:09:34,794 sllurp.llrp: INFO: will reset reader state on connect
2023-05-10 11:09:34,794 sllurp.llrp: INFO: will start inventory on connect
2023-05-10 11:09:34,794 sllurp.llrp: INFO: using antennas: [1]
2023-05-10 11:09:34,795 sllurp.llrp: INFO: transmit power: {1: 10}
2023-05-10 11:09:34,795 sllurp.llrp: INFO: connected to ('192.168.1.113', 5084) (192.168.1.113:5084)
2023-05-10 11:09:34,999 sllurp.llrp: INFO: using reader mode: None
2023-05-10 11:09:35,013 sllurp.llrp: INFO: stopping politely
Response:

Request:
"[{'EPC-96': b'393330393939393034353235', 'PeakRSSI': (-31,), 'LastSeenTimestampUTC': (1683708620491268,), 'TagSeenCount': (25,)}]"

2023-05-10 11:09:35,313 sllurp.llrp: INFO: Impinj search mode? None
2023-05-10 11:09:35,314 sllurp.llrp_proto: INFO: will report every ~N=50 tags
2023-05-10 11:09:35,314 sllurp.llrp: INFO: starting inventory
Response:

Request:
"[{'EPC-96': b'393330393939393034353235', 'PeakRSSI': (-31,), 'LastSeenTimestampUTC': (1683708621774632,), 'TagSeenCount': (50,)}]"
```
The new client connects to the reader.
\
As you are running Windows, your commands will be different, but the actions are the same.

fviard, 2025-03-18:
Closing this issue that is stale, but don't hesitate to re-open if needed.
Thank you @BjornGrylls for your reply that looks correct to me regarding the posted issue.

closed 2025-03-18 by fviard