sllurp/sllurp #179: [Question]How to set RX?

xang555, 2024-01-05:
How to set Rx to Impinj reader

fviard, 2024-01-06:
You mean, how to set the Rx antenna?
For Impinj, I don't think that the you can use different antenna for sending and for receiving.
The same antenna will be used for both.

xang555, 2024-01-06:
> You mean, how to set the Rx antenna? For Impinj, I don't think that the you can use different antenna for sending and for receiving. The same antenna will be used for both.

Thanks to reply 

I mean how to config RX value using this package? can you provide example to do that?

Thanks to advance

thijmenketel, 2024-03-29:
I don't exactly know how to configure the receiver sensitivity, however I remember that there's a way to extract the ReceiveSensitivityTableEntry from the reader. Unfortunately I can't find it anymore in this repo, it's still there on my (extremely dated) [fork](https://github.com/thijmenketel/sllurp/blob/0fe88eb2ddb262f54292cbdd5a145b0a2fec37d3/sllurp/llrp_proto.py#L1471). 

I do not have a reader to use anymore as I'm no longer at university so you'll have to try to find a solution yourself.

closed 2024-03-30 by xang555