sllurp/sllurp #188: Scanning Issues with Zebra Scanners

abhigya-maskay, 2024-11-05:
Scanning with the Zebra FX 9600 and FX 7500 is having issues and displaying some weird behavior. The devices connect and start the scan fine (and show no errors or warnings even with the debug mode on), however, only the first scan after a device reboot yields any tags at all. 

This behavior seem consistent with the reader being in autonomous mode as described here:
https://www.zebra.com/content/dam/support-dam/en/documentation/unrestricted/guide/software/interface-control-guide-en.pdf
under the `LLRP Custom Extensions Operation Section`

How would one go about writing custom llrp messages for the Zebra scanners to work with their extensions using the LLRPClient?

anonymous516, 2025-01-09:
I'm curious how you implemented custom LLRP messages using the LLRPClient to address this. Could you share any insights or details from your approach?

abhigya-maskay, 2025-01-15:
@anonymous516 This issue was originally asking how to add custom llrp messages. For testing I ended up using llrp-js for the custom messages. I havent had time to look at this particular issue in a bit but once I do I can update the thread

fviard, 2025-03-18:
Custom extensions are defined in `llrp_proto.py`.
For Zebra, there are all the message and parameters with name starting with Moto...

I think most Moto extensions are implemented on the "decode" side, ie to understand them, but not with "encoders" to be able to create messages using these extensions.

The autonomous mode implementation might be a little bit complicated, because it would require to rework the logic to be able to pass the "reader logic program" (ie ROSpec) at the initialization phase, and I'm not even convince that this mode has a real added value regarding the drawbacks that it brings.

But what we could probably do very easily, is use the extensions to "disable" the autonomous mode when connecting to the reader.

abhigya-maskay, 2025-03-18:
@fviard yes I would! Id be interested in getting these devices working so would be more than happy to test and provide anything required. Disabling the autonomous mode on connection sounds like it would fit my usage.

fviard, 2025-03-18:
@abhigya-maskay Awesome, I will work on the change and come back to you.