sllurp/sllurp #192: Implementing sllurp with FastAPI

KKP4, 2025-02-19:
I am implementing a small webserver to connect with the reader and send tags via HTTP.

I've established a connection with the reader and I can return tags via an endpoint on connect.

I'm trying to implement a start - stop reading mechanism via endpoint calls avoiding a "connect - join - disconnect" pattern.

Is there a way to do something like this:

- Establish connection to the reader on FastAPI boot
- call "GET/read" to start reading
- call "GET/stop" to stop reading and return the tags (but without disconnecting from the reader)

Also, I have a photo sensor on a GPI pin that should start the reading trigger when state is High. Is there any way to achieve this with sllurp?

Thanks in advance!

fviard, 2025-02-19:
I guess that you found the solution to your question as you closed the ticket but if ever you wanted to contribute a fastapi implementation for the samples in the example folder it would be great 😃

KKP4, 2025-02-19:
Yes, I found the solution in the /verbs folder :)
Start, stop, report endpoints are now working as intended.
I still need to test the GPI event trigger
I will submit my implementation when

fviard, 2025-02-19:
Thank you for the implementation.

I did not play with GPI events, but I think that there are 2 ways to do that:
1) The most efficient, but a lot more complicated that I will not recommend to you:
To control the reader, we create a sort of program with the configuration, this is what is called the "ROSpec".
Inside a ROSpec, you can have stop and start triggers, and GPI could be used as a trigger.
But that will be a little bit complicated because you might have to change a lot inside sllurp to do the ROSpec that you want.

2) The alternative that should not be too hard:
Before connecting, inside your config, you set to true "GPIEvent" of  "event_selector" of the config.
Then, you will indeed receive a callback event once GPI change of state.
And you could eventually start/pause reading. 
But, maybe, a more efficient solution would be to do that:
- Let the reader read tags continuously
- Discard card reads when you are not interested by them, and use an "event condition" or similar to indicate when not discarding tag reads
- Use the event callback to set up the event condition.

KKP4, 2025-02-21:
I have a FastAPI example PR ready. How do I submit it?

fviard, 2025-02-22:
Great, you should click on "fork" to create a fork of the project in your own repository.
Then, you can push your modification branch to your own repository, and github will show you options to do a "pull request" from the branch in your repository, to this repository.

closed 2025-03-18 by fviard