sllurp/sllurp #193: LLRP to TCP

tcccruz, 2025-02-20:
Hello, could you help me? I get the code to work fine. 
But I would like to make a server socket at the same time to output the tag id and its arrival time via TCP.
And at the same time and go recording a csv all those tags

Thank very much.

Jose Cruz

fviard, 2025-02-22:
You can find some examples to use sllurp as a library in the following folders:
- https://github.com/sllurp/sllurp/tree/develop/examples
- https://github.com/sllurp/sllurp/tree/develop/sllurp/verb

tcccruz, 2025-03-02:
Hello, thanks for the reply, I have it working now and everything is ok, the only problem is that I use the tornado example and I get a speed of about 6 Tags per second and instead the impinjtest application must be about 325000 per second.

fviard, 2025-03-08:
You need to set the proper (llrp) configuration for your test case and you should be able to get equivalent (or better) performance than with the Impinj test app.

For example, checking thinks like the mode, search mode, tag population estimation, because... this will impact the duration of "one round" of inventory and how long tags stay silent before replying again.

Also, you should ensure to use the report_every_n_tags=1 settings to ensure to get each read event immediately.