sllurp/sllurp #196: How to use sllurp to read the tags infinitely?

zhangzhehao996, 2025-03-05:
I use the mini example to read the tags, but it only read once and stuck by reader.join(), how to use the mini example to read the tags infinitely?

fviard, 2025-03-06:
By "mini example", you mean the code in the readme?

Maybe you can refer to the code of the "inventory" command that is a little bit more complete:
https://github.com/sllurp/sllurp/blob/develop/sllurp/verb/inventory.py

The tags and number of reads per tag that you will get is highly dependent of your configuration.

One config that might be interesting for you is:
report_every_n_tags=1
This will tell the reader to report every tag read event, instead of reporting all the reads only once you stop the reading.

Also, there might be a lot of RFID parameters to control the tags that might be important to have more tag "reads".
For example, the mode or search mode that you will use. As the tags might stay silent for some seconds after replying to reader request.

fviard, 2025-03-18:
@zhangzhehao996 Did you solve your problem and so we can close this issue? or can you provide more details?