sllurp/sllurp #200: Too few bytes (9) to unpack message header

vo1s, 2025-03-18:
Too few bytes (9) to unpack message header

I get this warning, please help. I think it doesn't work because of it

fviard, 2025-03-18:
Could you tell me which version of Sllurp that you are running? On which OS (linux, mac, ...) ? And which model of reader you are using?
Also, can you give me the command that you use to run sllurp?

Also, would it be possible for you to run the command with the "-d" flag at the beginning, and to paste the last lines, that should provide us with the debug info needed to understand your problem.
For example: `sllurp -d inventory  -X .....`

vo1s, 2025-03-20:
I am usin sllurp     2.0.1 on windows chainway ur4
I use "sllurp inventory 192.168.99.202" command